TheThingsIndustries/lorawan-stack-docs #17: Local env setup doesn't work

KrishnaIyer, 2020-10-05:
#### Summary

Local env setup doesn't work (aka the `make` commands)

#### Steps to Reproduce

1. Download Repo
2. Run `$ make init`
3. Run `$ make server`

#### What do you see now?

```
> /Users/krishnaiyereaswaran/.yarn already exists, possibly from a past Yarn install.
> Remove it (rm -rf /Users/krishnaiyereaswaran/.yarn) and run this script again.
yarn --cwd doc/themes/the-things-stack/
make: yarn: No such file or directory
make: *** [doc/themes/the-things-stack/node_modules] Error 1

```

I already have `yarn` installed for `lorawan-stack`. I shouldn't need to remove existing installation for this to work.

#### What do you want to see instead?

Hugo serving docs locally.

#### Environment

`macOS Catalina 10.15.7`

#### How do you propose to implement this?

No clue

#### How do you propose to test this?

`$ make server` , `$ make deps` etc work without errors

#### Can you do this yourself and submit a Pull Request?

Can test

benolayinka, 2020-10-06:
I can't reproduce this

```
ben$ git clone git@github.com:TheThingsIndustries/lorawan-stack-docs.git test
Cloning into 'test'...
Bens-MBP:~ ben$ cd test/
Bens-MBP:test ben$ make init
go run .hooks/install-hooks.go
Installing commit-msg hook
yarn --cwd doc/themes/the-things-stack/
yarn install v1.22.4
[1/4] 🔍  Resolving packages...
[2/4] 🚚  Fetching packages...
[3/4] 🔗  Linking dependencies...
[4/4] 🔨  Building fresh packages...
✨  Done in 0.24s.
Bens-MBP:test ben$ make server
go run -tags extended github.com/gohugoio/hugo server -s doc --environment gh-pages
# github.com/bep/golibsass/internal/libsass
In file included from inspect.cpp:2:
../go/pkg/mod/github.com/bep/golibsass@v0.6.0/internal/libsass/../../libsass_src/src/inspect.cpp:1069:20: warning: expression with side effects will be evaluated despite being used as an operand to 'typeid' [-Wpotentially-evaluated-expression]
port 1313 already in use, attempting to use an available port
Building sites … WARN 2020/10/06 17:40:27 .File.Ext on zero object. Wrap it in if or with: {{ with .File }}{{ .Ext }}{{ end }}

                   | EN   
-------------------+------
  Pages            | 216  
  Paginator pages  |   0  
  Non-page files   | 224  
  Static files     |  26  
  Processed images |   0  
  Aliases          |  40  
  Sitemaps         |   0  
  Cleaned          |   0  

Built in 1027 ms
Watching for changes in /Users/ben/test/doc/{content,data,layouts,static,themes}
Watching for config changes in /Users/ben/test/doc/config/_default, /Users/ben/test/doc/config/gh-pages
Environment: "gh-pages"
Serving pages from memory
Running in Fast Render Mode. For full rebuilds on change: hugo server --disableFastRender
Web Server is available at http://localhost:62749/ (bind address 127.0.0.1)
Press Ctrl+C to stop
```

benolayinka, 2020-10-06:
it's probably here in the makefile

```
$(YARN_DEPS):
	@if ! [ -x "$$(command -v yarn)" ]; then\
		echo "Installing yarn";\
	    curl -o- -L https://yarnpkg.com/install.sh | bash;\
	fi
	yarn --cwd doc/themes/the-things-stack/
```

do you have `yarn` in $PATH somewhere? If you don't have it and then it fails to install, I guess that could cause your error

closed 2020-11-10 by benolayinka