TheThingsIndustries/lorawan-stack-docs #46: Docs reorganization umbrella issue

benolayinka, 2020-10-22:
#### Summary

Nejra suggested a great list of docs improvements.

- Don't use "Enterprise" in value statements, it's confusing because of the distribution names
- Same with "Open source" 
- Visually align tables in Feature Comparison
- Restructure section order - think application, then gateway, then device
- Create quick start
- Improve distribution marking
- Users and Events both require an application - restructure so these are encountered in correct order
- Console before CLI
- Since we provide example files in installation guide, mark the explanation as optional.
- Note not to change TLS source in example installation files
- Troubleshooting is a living doc
- Migrating devices API chart can be improved - what is "not in documentation?
- Reorganize device section slightly (see presentation)
- Limit height of tall pictures
- Remove "adding integrations" card
- Improve sidebar
- Use collapse for code
- Use a better code viewer like https://cloud.google.com/functions/docs/quickstart#functions-clone-sample-repository-nodejs

#### Can you do this yourself and submit a Pull Request?

Will work on it!

johanstokking, 2020-11-12:
This is on-going, right?

Can you verify the milestone?

benolayinka, 2021-05-04:
With The Things Stack conf end of this month, we should think about restructuring docs a little for newcomers. Proposing a structure something like this:

Top level sections
- Getting started
- Devices and gateways
- Integrations
- Reference
- Troubleshooting

Getting started
- quick start
- what is the things stack?
- installing the cli
    - adding devices
    - adding gateways
    - configuring mac settings
    - enabling device classes
    - downlink queue ops
    - working with events
    - examples for everything
- installing the things stack
- migrating to the things stack
- managing users and organizations
- aws launcher
- cloud
- ttn

Devices and gateways
- adding applications
- adding devices
- adding gateways
    - all content from gateways section
- best practices
- device claiming & qr codes
- end device templates
- mac settings
- multicast

Integrations
- Using the API
- Webhooks
- MQTT
- AWS IoT
- LoRa Edge
- Cloud integrations
- Payload formatters
- Storage integration

Reference
- Billing with Strip
- API
- Application Packages
- CLI
- Components
- Data Formats
- Email Templates
- Federated Auth
- Frequency Plans
- Glossary
- Everything from reference

Troubleshooting
- Consolidate all troubleshooting sections

@nejraselimovic @laurensslats what do you think?

nejraselimovic, 2021-05-05:
I think that putting sections like adding devices, gateways, etc. under Installing the CLI would just introduce repetition and confusion. Adding applications should remain under Integrations (I checked that with Laurens before), but in general, a joint Devices and Gateways section might be a good idea.  I'm also not sure about consolidating troubleshooting sections, that might become too complicated at some point.

In general, I'm happy with what we have now, and I'm not sure if this change would bring much difference.

johanstokking, 2021-05-07:
I'm also pretty happy with the structure that we have. I would indeed keep pages for an "action" (e.g. add application, send downlink message), and on that page, put alternatives for Console, CLI (and API).

That said, I do have a few suggestions that are low hanging fruit:

1. Integrations: place "Integrations with MQTT Clients" under "MQTT Server"
2. Remove "Integration"; we're already in the section Integrations. So "Node-RED", "IFTTT". The exception might be "Storage Integration"
3. Move "AWS IoT" under "Cloud Integrations"
4. Move "Visualize LoRa Edge Gelocation" to _under_ Ubidots, as an example
5. Document "LoRa Cloud" under "Cloud Integrations". That should include Console screens. And it supersedes https://www.thethingsindustries.com/docs/reference/application-packages/lora-cloud-device-and-application-services/

nejraselimovic, 2021-05-07:
> I'm also pretty happy with the structure that we have. I would indeed keep pages for an "action" (e.g. add application, send downlink message), and on that page, put alternatives for Console, CLI [...]

Those are really good suggestions, I'll work on that in the following weeks

PR #374 (2021-06-10): Docs reorganization
#### Summary
Final docs reorganization to close #46

#### Changes
- Moved AWS IoT to Cloud Integrations
- Moved LoRa Edge data visualization via Ubidots under Ubidots as an example 
- Moved Integrations with MQTT Cliends under MQTT Server
- Changed the order of sections in Getting Started and Devices, because it's not really logical. For example, Troubleshooting Getting Started was somewhere in the middle of left hand menu, instead of at the end of the list. One more example - I want to see Console (how to log in) after Installing The Things Stack, because you simply cannot [... cut]
closed 2021-06-10 by johanstokking