TheThingsIndustries/lorawan-stack-docs #79: Download page problems

michalborkowski96, 2020-11-11:
1. Link 'Download' to open source version is broken.
2. First (and last) row of the table doesn't look good on certain screens (see image). Words are broken where they shouldn't and buttons aren't in one line.

[image]

@benolayinka

johanstokking, 2020-11-12:
Need milestone

johanstokking, 2020-11-12:
@michalborkowski96 these are two separate issues.

@benolayinka please fix at least the downlink page as part of this milestone.

benolayinka, 2020-12-30:
Download link was fixed in #167 .
Download page will be replaced by https://thethingsindustries.com/technology/pricing/ or something similar

closed 2020-12-30 by benolayinka