TheThingsIndustries/lorawan-stack-docs #86: LR1110 Lora Cloud Integration

benolayinka, 2020-11-12:
#### Summary

Documentation for the LoRa Cloud Integration should be in Integrations. It's currently in reference/application-packages.

#### How do you propose to document this?

Integrations/LoRa-Cloud with a guide about using it, and maybe more technical API stuff can stay in reference/application packages (probably not necessary though.)

Also take a look at the storage integration package, these packages should have the same directory structure and organization.

#### Can you do this yourself and submit a Pull Request?

@nejraselimovic w/ @adriansmares

johanstokking, 2020-11-12:
Need milestone

nejraselimovic, 2020-12-18:
@benolayinka can I just add a **LoRa Cloud Integration** section and point to this https://www.thethingsindustries.com/docs/integrations/lr1110-geolocation/ example?

benolayinka, 2020-12-21:
> @benolayinka can I just add a **LoRa Cloud Integration** section and point to this https://www.thethingsindustries.com/docs/integrations/lr1110-geolocation/ example?

This should be closed by the LR1110 integration you wrote. All done!

benolayinka, 2020-12-21:
Closed by #165

closed 2020-12-21 by benolayinka