TheThingsIndustries/lorawan-stack-docs #92: Document RAK 7240

benolayinka, 2020-11-16:
#### Summary

Document connecting RAK 7240

#### Why do we need this ?

Document popular gateways

benolayinka, 2021-02-25:
@nejraselimovic can you pick this up? i'll send RAK a message for their documentation, it should just be a matter of converting it to a markdown doc

nejraselimovic, 2021-03-01:
> @nejraselimovic can you pick this up? i'll send RAK a message for their documentation, it should just be a matter of converting it to a markdown doc

Sure, tell me when you have it

descartes, 2021-03-03:
https://docs.rakwireless.com/Product-Categories/WisGate/RAK7240/Overview/

benolayinka, 2021-03-03:
> > @nejraselimovic can you pick this up? i'll send RAK a message for their documentation, it should just be a matter of converting it to a markdown doc
> 
> Sure, tell me when you have it

RAK never responded. You can probably just put together something based on this document -> https://docs.rakwireless.com/Knowledge-Hub/Learn/WEB-Management-Platform/#basic-station

Specify instructions for Basic Station, based on our Basic Station docs.

i3mSuperMan, 2021-03-10:
@benolayinka,

We believe connecting below RAK Model Gateways with the TTSC process is the same.

- RAK7258
- RAK7249
- RAK7240

Ref: https://docs.rakwireless.com/Knowledge-Hub/Learn/WEB-Management-Platform/#basic-station

It would be helpful if you can add **NOTE** on `RAK7258` and `RAK7249` configurations in the `RAK7240` guide.

closed 2021-03-31 by nejraselimovic