TheThingsIndustries/lorawan-stack-docs #102: Document usage of device location in Console

benolayinka, 2020-11-17:
#### Summary
Moved from https://github.com/TheThingsNetwork/lorawan-stack/issues/3124

It's possible to add a location to a device, but it's not clear how/when this location is used and if this location can be set via other ways than adding the lat and lon in the console.

[image]

#### Why do we need this ?
It's not clear how this feature can be used. 

#### What is already there? What do you see now?
See img 👆 
Setting location via console.

#### What is missing? What do you want to see?
Documentation on:
- What are the sources for getting this location (only console, or also via the GPS coordinates of the devices, or via another way...) 
- If this is set, how can it be used? Only for reference in the console, or it is also added when initiating integrations?

#### How do you propose to document this?
1. Find out details from @htdvisser 
2. Write documentation by @benolayinka

htdvisser, 2020-11-17:
I don't have details for this.

benolayinka, 2020-11-18:
I think @neoaggelos or @kschiffer know how this works

kschiffer, 2020-11-18:
We have different sources for the location data, one of them being the registry, which is also what the console is using to update the location. However, there are also other sources that the location can come from:

```protobuf
var LocationSource_name = map[int32]string{
	0: "SOURCE_UNKNOWN",
	1: "SOURCE_GPS",
	3: "SOURCE_REGISTRY",
	4: "SOURCE_IP_GEOLOCATION",
	5: "SOURCE_WIFI_RSSI_GEOLOCATION",
	6: "SOURCE_BT_RSSI_GEOLOCATION",
	7: "SOURCE_LORA_RSSI_GEOLOCATION",
	8: "SOURCE_LORA_TDOA_GEOLOCATION",
	9: "SOURCE_COMBINED_GEOLOCATION",
}
```

From what I understand, when the gateway/device has such functionality activated, it can (over-)write the location automatically (e.g. based on uplink metadata). In that case, what users would see in the Console is not what they originally entered, but the location that was set automatically by the NS/GS. We currently don't communicate where the location data is actually coming from in the Console, which should be addressed, fwiw.

Maybe @neoaggelos and/or @rvolosatovs can tell you some more about how non-registry locations work for devices/gateways.

Also note that gateways can set a location per antenna, however, in the Console we currently only support setting (and showing) the location for the first antenna.

As far as the Console is concerned, we only use location to display it in the map widget. But I also understand that location data for gateways can for example be used to triangulate the position of an end device, which in turn is what

```
	7: "SOURCE_LORA_RSSI_GEOLOCATION",
	8: "SOURCE_LORA_TDOA_GEOLOCATION",
	9: "SOURCE_COMBINED_GEOLOCATION",
```

is meant for.

rvolosatovs, 2020-12-19:
@kschiffer did a great job explaining how this works in https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/102#issuecomment-729668168
perhaps @KrishnaIyer can give some details on gateway-level specifics here

KrishnaIyer, 2020-12-22:
> From what I understand, when the gateway/device has such functionality activated, it can (over-)write the location automatically (e.g. based on uplink metadata). In that case, what users would see [...]

`SOURCE_GPS`: This is only for gateways that support GPS and are using authentication. 
- UDP gateways support GPS information but connects insecurely. So we don't allow those gateways to update locations.
- Basic Station protocol does not yet support GPS coordinates in its messages.
-  ATM only MQTT gateways with GPS can actually update these fields, which is a very tiny fraction of the gateways.

`SOURCE_IP`: Afaik this is based on the IP address of the gateway and is quite unreliable. 

> Also note that gateways can set a location per antenna, however, in the Console we currently only support setting (and showing) the location for the first antenna.

The GS itself does not have full support for multiple antennas yet. We have an [issue](https://github.com/TheThingsNetwork/lorawan-stack/issues/2006) for that in backlog.

nejraselimovic, 2020-12-23:
So once the location is obtained (through any of those sources), at this moment it's not being used anywhere besides the map widget in the Console?

nejraselimovic, 2020-12-23:
> `SOURCE_GPS`: This is only for gateways that support GPS and are using authentication.
>
>     * UDP gateways support GPS information but connects insecurely. So we don't allow those gateways to update [...]

I tried using `ttn-lw-cli` with `--antenna.location.source` and whatever I provide as a source, it returns `SOURCE_REGISTRY`. Why is this?

benolayinka, 2020-12-28:
> ```go
> var LocationSource_name = map[int32]string{
>     0: "SOURCE_UNKNOWN",
>     1: "SOURCE_GPS",
>     3: "SOURCE_REGISTRY",
>     4: "SOURCE_IP_GEOLOCATION",
>     5: "SOURCE_WIFI_RSSI_GEOLOCATION",
>     6: "SOURCE_BT_RSSI_GEOLOCATION",
>     7: "SOURCE_LORA_RSSI_GEOLOCATION",
>     8: "SOURCE_LORA_TDOA_GEOLOCATION",
>     9: "SOURCE_COMBINED_GEOLOCATION",
> }
> ```

How can users enable any of these?

nejraselimovic, 2021-01-13:
Closed by #176

closed 2021-01-13 by nejraselimovic