TheThingsIndustries/lorawan-stack-docs #95: Create `whats-new/v3.x`

benolayinka, 2020-11-17:
#### Summary

Migrated from https://github.com/TheThingsIndustries/lorawan-stack/issues/1681#event-3988842893

Add a `whats-new-in-minor` section to docs, which we copy over from Changelog.

#### Why do we need this ?

Share what's new in docs

#### Can you do this yourself and submit a Pull Request?

I can create, but product owners need to populate it

benolayinka, 2020-11-17:
@johanstokking do we want to update these files as part of new feature PRs, or once per release?

Also, how do we display them? Maybe the latest what's new on the homepage? A separate what's new page with all of the version updates?

johanstokking, 2020-11-17:
It would be per release.

Note that November 2020 is frozen, moving to December.

closed 2021-04-20 by benolayinka