TheThingsIndustries/lorawan-stack-docs #106: Document that pub/subs are deprecated

johanstokking, 2020-11-18:
#### Summary

We are deprecating pub/subs in The Things Stack Cloud and Enterprise.

Raised priority because this needs to be done asap.

#### Why do we need this ?

We should inform users that they should not use pub/sub for new use cases.

#### What is already there? What do you see now?

Documentation on configuring pub/sub for an external MQTT server and NATS broker.

#### What is missing? What do you want to see?

Deprecation warning that MQTT and NATS pub/sub this will be disabled on The Things Stack Cloud from January 5th 2021, and that people should use webhooks instead.

#### How do you propose to document this?

Add a short code, because we probably want to deprecate things in the future. This would be for a distribution and a date. Probably as big as the CLI only short code.

#### Can you do this yourself and submit a Pull Request?

Can review

benolayinka, 2020-11-18:
@nejraselimovic this one should be pretty easy, wanna pick it up?

nejraselimovic, 2020-11-18:
> @nejraselimovic this one should be pretty easy, wanna pick it up?

Sure, I'll pick it up right now

closed 2020-12-02 by nejraselimovic