TheThingsIndustries/lorawan-stack-docs #114: Write Upgrading Guide

htdvisser, 2020-11-19:
#### Summary

We should write a guide that explains operators how to upgrade The Things Stack.

#### Why do we need this ?

Because operators need to understand what to do with minor / patch releases.

#### What is already there? What do you see now?

Only deployment instructions.

#### What is missing? What do you want to see?

Instructions for updating to a new minor / patch release.

#### Can you do this yourself and submit a Pull Request?

Yes, but my TODO list is pretty full, so it will have to wait.

michalborkowski96, 2021-04-22:
With #115 closed, can you tell me in more detail what can I do regarding this issue?

htdvisser, 2021-04-30:
https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/115 only documented the process for ECS deployments, and does not cover other self-hosted deployments (using Docker-Compose, or custom Docker/Systemd). 

This issue is about high-level upgrading documentation, so things like:

- Pushing front-end assets to your CDN (if you use it)
- Stopping the deployment
- Running before-upgrade data/database migrations
- Replacing The Things Stack (updating image / replacing binary and assets)
- Restarting the deployment
- Running after-upgrade data migrations (if upgraded without stopping the deployment)
- Monitoring

michalborkowski96, 2021-04-30:
Uhm... I've only dealt with the AWS ECS deployment, so...

NicolasMrad, 2022-03-29:
is this still relevant? It has been been moved for 3 quarters

htdvisser, 2022-04-11:
I think it is, but it seems I'm the only one. So I'm fine with closing this until there's actual demand from users.

closed 2022-04-11 by htdvisser