TheThingsIndustries/lorawan-stack-docs #122: Document API examples for Storage Integration

benolayinka, 2020-11-24:
#### Summary

Add examples for how to access storage integration data with different languages.

See https://github.com/TheThingsIndustries/lorawan-stack-support/issues/223#issuecomment-732978256

#### Why do we need this ?

Make storage integration easier to use, and ease transition from v2 to v3

#### What is already there? What do you see now?

Basic ttn-lw-cli commands.

#### What is missing? What do you want to see?

Python, go, javascript, curl examples.

#### How do you propose to document this?

@neoaggelos

neoaggelos, 2020-11-28:
A problem I see here is that most of the code of the examples is applicable to other APIs as well, and using TTS programmatically is something that is currently lacking in the documentation.

I would consider breaking down this issue into separate ones:
- Add generic documentation code examples for using TTS APIs from Python/Go/Javascript/curl (for curl there is #28). This should be akin to https://www.thethingsnetwork.org/docs/applications/sdks.html
- Add a separate section for handling streaming APIs that use server-sent events. This is because subscribing to TTS events work the same way, and so it is not specific to the storage integration API.
- For the storage integration, add a simple example, and point to the code examples documentation for more generic stuff (e.g. authentication, URLs)

I think that could be a good solution and also provide a good resource for people looking to develop applications on top of TTS.

neoaggelos, 2020-12-24:
Will do after https://github.com/TheThingsIndustries/lorawan-stack/issues/2453

johanstokking, 2021-03-01:
@neoaggelos what's the status here?

descartes, 2021-03-03:
I have some sample code for MQTT, WebHooks and Data Storage that I'm morphing in to various deployments - desktop app, web pages, SPA etc.

Because Server Sent Events is CORS sensitive, I've tried figure the code from the console as you don't appear to be using SSE to subscribe to them but tracing how you do got too much of a time sink, so an example for that would be useful.

The Data Storage is delivered as chunks like SSE which needs a bit of pre-processing if you just used XHR to fetch data - maybe this could be split.

johanstokking, 2021-03-04:
We don't want a browser example for using the Storage Integration, because that means that an API lives in the browser. That is not a real world scenario that we should document as a reference.

This issue is also not about MQTT or webhooks, but about the Storage Integration. Let's focus on that and keep it small and simple. We just need simple code examples of how to use the Storage Integration as a client, in Node, Python and Go to start with.

neoaggelos, 2021-03-18:
#186 was closed because we decided that it is best to have example code in a separate repository rather than the docs.

@htdvisser I do not have permission to create a new repository, can you create `TheThingsNetwork/lorawan-stack-examples` so that I can start adding code examples?

htdvisser, 2021-03-19:
created

neoaggelos, 2021-08-17:
We already have Python and Go examples in https://github.com/ThEThingsIndustries/lorawan-stack-examples.

The default branch in that repository is `go-examples`, but it should be changed to `master`.

johanstokking, 2021-08-18:
OK so what is the plan here; are we mixing Python and Go and potentially multiple languages in one repo? Why?

Do we start with examples for receiving webhooks?

closed 2022-02-03 by NicolasMrad