TheThingsIndustries/lorawan-stack-docs #139: Inline shortcodes in section headers break anchor links

benolayinka, 2020-12-02:
#### Summary

Adding shortcodes to section headers changes the section id and therefore breaks anchor links

[image]
[image]

#### Can you do this yourself and submit a Pull Request?

Yes

benolayinka, 2020-12-02:
Solutions I see are to

- Do something clever to filter shortcodes when generating the id slug
- Wrap the header in a shortcode
- Move the shortcode to another line

@htdvisser what do you think?

htdvisser, 2020-12-07:
I think the "Cloud and Enterprise Only" stuff could be moved to a different line. I think we already had an idea about using "tags" to indicate applicable distributions, then we can just as well add some CSS to make them float on the right of the section.

Another option would be to wrap the entire section with a shortcode, then we can even add a nice colored line on the right to visualize where that "Cloud and Enterprise Only" part ends.

PR #162 (2020-12-16): Quickfix for inline shortcodes breaking headings

#### Summary

Fixes #139 until we find a better solution

#### Screenshots

[image]
[image]

#### Changes

- Add CONTRIBUTING note about manually fixing headings with inline shortcodes
- Fix existing headings with inline shortcodes

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, a [... cut]
closed 2020-12-16 by benolayinka