TheThingsIndustries/lorawan-stack-docs #141: Improve documentation Storage Integration

laurensslats, 2020-12-02:
#### Summary
I received questions from a customer who's not able to set up the storage integration, neither am I.

#### Why do we need this ?
Improve user experience

#### What is already there? What do you see now?
https://thethingsstack.io/integrations/storage/configuration/

https://thethingsstack.io/reference/configuration/application-server/#storage-integration-options-hahahugoshortcode-s1-hbhb

#### What is missing? What do you want to see?
1. Explain, or provide a link to a section where we explain what `ttn-lw-stack.yml` is, where to find it and how to edit this file?

In the documentation we only explain how to install this

[image]

2. The anchor link **Storage Integration Options** doesn't work.

It links to 
`https://thethingsstack.io/reference/configuration/application-server/#storage-integration-options`
the right link to use is:
`https://thethingsstack.io/reference/configuration/application-server/#storage-integration-options-hahahugoshortcode-s1-hbhb`

[image]

3. [...application-server/#storage-integration-options](https://thethingsstack.io/reference/configuration/application-server/#storage-integration-options-hahahugoshortcode-s1-hbhb) is a complete mystery to me. Don't expect people to understand any of it. Either remove the link in https://thethingsstack.io/integrations/storage/configuration/ or explain this is an advanced options so people don't mind that they don't understand.

#### How do you propose to document this?
Update docs.

#### Can you do this yourself and submit a Pull Request?
Needs input from @benolayinka and @neoaggelos

TheThingsBot, 2020-12-02:
This issue has been mentioned on **Discuss The Things Industries**. There might be relevant details there:

https://discuss.thethingsindustries.com/t/data-storage-tts-cloud-im-puzzled/403/1

benolayinka, 2021-01-12:
Closed by https://github.com/TheThingsIndustries/lorawan-stack-docs/commit/97fba03d34530463f12a6c84143c47ad6bf3d006

closed 2021-01-12 by benolayinka