TheThingsIndustries/lorawan-stack-docs #146: Add documentation for bulk device claiming via CLI

ajovertoom, 2020-12-04:
#### Summary

Offer our customers to bulk claim devices via CLI or console.

#### Why do we need this ?

Better UX and more efficient ops for customers.

#### What is already there? What do you see now?

Per device claiming with QR-code via separate page <example>.eu1.cloud.thethings.industries/claim: https://thethingsstack.io/devices/device-claiming/claim-devices/

#### What is missing? What do you want to see?

- Provide documentation/scripts on how to use CLI to claim devices in bulk and add here https://thethingsstack.io/devices/device-claiming/claim-devices/
- Provide `Claim device` option in console: next to `Import end devices` (or perhaps as part of `Import end devices`) that allows for manual or JSON (?) entry of identifiers (joinEUI, devEUI, claim authentication code), : https://wilhelmsen.eu1.cloud.thethings.industries/console/applications/digimatter-test/devices

#### How do you propose to document this?

Follow what's currently there: https://thethingsstack.io/devices/device-claiming/claim-devices/

#### Can you do this yourself and submit a Pull Request?

Nope need @mjamescompton and @kschiffer

benolayinka, 2021-01-04:
Also had some feedback requesting this. Adding prio/medium.

kschiffer, 2021-06-01:
Please open a new issue about the Console implementation in [https://github.com/TheThingsNetwork/lorawan-stack](https://github.com/TheThingsNetwork/lorawan-stack) if needed. Unassigning myself.

benolayinka, 2021-07-12:
who wrote the cli functionality? @neoaggelos ?

closed 2026-10-08 by mjamescompton