TheThingsIndustries/lorawan-stack-docs #160: Multiple docs with same title breaks search

benolayinka, 2020-12-15:
#### Summary

Every cloud integration has a section called creating webhooks, and this confuses search.

Possible solutions
- Rename the sections to include the parent section name (i.e creating a webhook for Akenza)
- Linking to one generic section

#### Can you do this yourself and submit a Pull Request?

@nejraselimovic putting you in charge

ajovertoom, 2020-12-16:
Also beyond the integrations, this applies to Basic Station, see scrshot. And there are probably more generic concepts with multiple bespoke implementations.
[image]

nejraselimovic, 2020-12-16:
I would go for renaming the sections

nejraselimovic, 2020-12-16:
I'm not sure how to approach this. I was thinking that the most easy way was to just rename the sections, but look at the following example. 

[image]

If you look at the last part **Migrating to The Things Stack**, clicking the first result leads [here](https://thethingsindustries.com/docs/getting-started/migrating/) which is fine, but the second results leads [here](https://thethingsindustries.com/docs/getting-started/migrating/configure-mac-settings/) which is not fine, and as far as I can see the problem there isn't the section name. I don't have a clue how this search works, so I'm not sure if renaming sections would resolve this issue entirely.

benolayinka, 2020-12-16:
> I'm not sure how to approach this. I was thinking that the most easy way was to just rename the sections, but look at the following example.
>
> [image]
>
> If you look at the last part **Migrating to The [...]

Renaming the sections is a good start, let's do what we can to fix that. The Migrating issue is weird, I'll message Algolia about it.

nejraselimovic, 2020-12-16:
> Renaming the sections is a good start, let's do what we can to fix that. The Migrating issue is weird, I'll message Algolia about it.

Deal, I'm on it tommorow

PR #166 (2020-12-23): Issue/rename sections
#### Summary
Renaming sections to resolve #160 and some minor updates

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [x] Style Guidelines: Documentation obeys style guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [x] Commits: Commit messages  [... cut]
closed 2020-12-23 by nejraselimovic