TheThingsIndustries/lorawan-stack-docs #209: Improve TTS Architecture documentation

KrishnaIyer, 2021-01-26:
#### Summary

Improve TTS Architecture documentation

#### Why do we need this ?

A lot of people are asking the same questions.

#### What is already there? What do you see now?

Just a simple components view somewhere hidden in the docs https://www.thethingsindustries.com/docs/reference/components/

#### What is missing? What do you want to see?

Make it better

#### How do you propose to document this?

1. Make the TTS Architecture more prominent in the docs by making it the second item on Getting Started.
 -  Add a better architectural overview of the components (like zo)
[image]

2. Add some flows
[image]

> Note: Images from my Masters Thesis.. xD

#### Can you do this yourself and submit a Pull Request?

Yes

descartes, 2021-03-03:
Add some flows - oh yes please - particularly if it points to the code module in the repro

Can you enable the Thumbs Up button so we can 'vote' please

NicolasMrad, 2022-03-29:
is this still relevant?

KrishnaIyer, 2022-03-30:
Yup.

KrishnaIyer, 2022-06-10:
Haven't really had a chance. Since this doesn't seem so relevant, I'm going to remove this from a milestone. Based on more upvotes, I can pick this up.