TheThingsIndustries/lorawan-stack-docs #210: Clarify V2 to V3 migration guide

johanstokking, 2021-01-28:
#### Summary

Clarify V2 to V3 migration guide.

See https://www.thethingsnetwork.org/forum/t/ttn-v2-to-the-things-stack-v3-transition/43247/36

#### Why do we need this ?

Because apparently there are too many assumptions in the current documentation.

#### What is already there? What do you see now?

https://www.thethingsindustries.com/docs/getting-started/migrating/migrating-from-v2/

#### What is missing? What do you want to see?

1. Do explain the Windows version as alternative steps. This is still the most popular desktop OS. Will `ttnctl selfupdate` work or is it `ttnctl.exe selfupdate`? Those small things
2. People want to follow instructions and expect that they are in this guide. It's very easy to miss the reference to the instructions for The Things Network. People will likely follow the instructions on this page, and mistakenly try to configure `ttnctl` for TTN with the V2 private network instructions
3. It is not clear what to do after exporting. There is no link in the bottom that one should now import that file somewhere

#### How do you propose to document this?

1. Make the instructions for Windows, macOS and Linux of equal importance
2. Make the guide self-containing. Better not refer to the old TTN docs (we get rid of those anyway). If you want to present two alternatives (TTN V2 vs private V2), put them in tabs like we do with Open Source vs Enterprise
3. Link in the bottom to https://www.thethingsindustries.com/docs/getting-started/migrating/import-devices/

#### Can you do this yourself and submit a Pull Request?

I'm off next week, please monkey test this and merge soon, people are already stuck.

johanstokking, 2021-01-28:
Also why are we using `ttnctl` in the docs and not `ttn-lw-migrate`? That is the preferred way, right?

I'm now looking at the migration session on The Things Conference (https://app.hopin.com/events/the-things-conference-2021/stages/a901d5a9-1f8b-4532-a5a5-d0d486c9d388) and that looks much easier that way.

nejraselimovic, 2021-01-28:
I think @neoaggelos first needs to update the documentation if needed (for example using `ttn-lw-migrate` instead of `ttnctl`), and I can do testing on Ubuntu and Windows, then try to clarify the documentation as much as possible.

johanstokking, 2021-01-28:
cc @ymgupta

johanstokking, 2021-01-28:
> I think @neoaggelos first needs to update the documentation if needed (for example using `ttn-lw-migrate` instead of `ttnctl`), and I can do testing on Ubuntu and Windows, then try to clarify the documentation as much as possible.

OK, @neoaggelos @ymgupta please request @nejraselimovic for review.

neoaggelos, 2021-01-28:
There is https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/130

johanstokking, 2021-01-28:
@neoaggelos you own #130 so please keep moving with that.

Also it's not entirely addressing this issue, but it's a good start.

laurensslats, 2021-05-20:
@nejraselimovic @neoaggelos pls have a look at this issue

nejraselimovic, 2021-06-09:
I think this can be closed now as we have made a full reorganization and update of migration docs.

closed 2021-06-09 by nejraselimovic