TheThingsIndustries/lorawan-stack-docs #213: Improve navigation

johanstokking, 2021-01-29:
#### Summary

Improve navigation.

I would personally prefer having a structured, large menu, on the left, that is constantly visible.

#### Why do we need this ?

To provide a better overview of the documentation structure and improve navigation for users.

#### What is already there? What do you see now?

With the tiles currently, it's confusing in which section of the documentation you are, especially if you're halfway through. It also costs a lot of screen space, and you have to go left and right with your eyes all the time. Also, since not everything is sorted, it's pretty challenging to find things, even if you expect them to be there.

#### What is missing? What do you want to see?

I would like to see a permanent menu on the left with the table of contents.

For reference:
- https://doc.sm.tc/station/
- https://docs.edgeimpulse.com/docs/getting-started
- https://docs.influxdata.com/influxdb/v1.8/
- https://docs.docker.com/get-started/ (they do have a nice landing page here: https://docs.docker.com/)

#### How do you propose to implement this?

We probably have to rework the theme for this. I suppose that the existing documentation contents can be left in tact. We probably also need that not to break links.

#### Can you do this yourself and submit a Pull Request?

Can review

johanstokking, 2021-01-29:
@kschiffer please share thoughts on this as well

nejraselimovic, 2021-02-03:
Ben and I have been planning to work on this for quite some time now, but I've been unable to sort it out on my own. @benolayinka could you take this over?

kschiffer, 2021-02-03:
> With the tiles currently, it's confusing in which section of the documentation you are, especially if you're halfway through. It also costs a lot of screen space, and you have to go left and right [...]

Agreed. Especially the issue with wasting screen space, which is already a problem when displaying tables. I can offer to take over the implementation after I have made some checkmarks on TTS. @benolayinka and @nejraselimovic, maybe it's good if we set up a meeting then so we can discuss some improvement points. Some that I see are (heavily inspired by docker docs):

*   Improve navigation
    *   Add a permanent sidebar with all topics for all content pages
    *   Modify the layout to make more efficient use of screen space
        *   Shrink the side-bar size, move to left (completely)
        *   Add nested elements
        *   Maybe add option to minimize sidebar
    *   Add quick-actions to every content page (edit this page, jump to sections)
    *   Make sections linkable (via hashes)
*   Other things that are mildly related to navigation
    *   Improve landing page a bit
        *   The current hero is too much marketing, people will visit the docs to get info on specific topics, not to be advertised
        *   Based on the search statistics, feature common topics prominently
        *   Make download button a bit clearer
        *   Fix SVG links below the hero (you have to click right on the illustration)
    *   Rework typography spacings and scale

It could be challenging to set this up in a way that keeps the links intact, but I'll have to take a closer look at this.

Also, I'd prefer getting rid of bulma, which repeatedly forces us into design compromises.

benolayinka, 2021-02-03:
@kschiffer @johanstokking i was planning to borrow this theme, since it's already written for hugo https://doks.netlify.app/docs/prologue/introduction/

kschiffer, 2021-02-03:
Yeah, this one looks like it already solves a lot of problems. But one has to see how it looks with our content. Regarding the sidebar, we might need to modify this to be expandable/collapsable, because otherwise the sidebar might become too long.

closed 2021-03-08 by benolayinka