TheThingsIndustries/lorawan-stack-docs #218: Title anchor with "new in version" tag not rendering correctly inside tabs

nejraselimovic, 2021-02-04:
#### Summary
When a title is inside tabs, the anchor doesn't render correctly if there's also a `new  in version` tag.

#### What is already there? What do you see now?
[image]
[image]

#### What is missing? What do you want to see?
Works well if outside of tabs.

nejraselimovic, 2021-03-24:
Another thing relatable to using tabs -> right hand navigation not working on this page

[image]

benolayinka, 2021-07-12:
Closed by a commit I can't find, but https://github.com/TheThingsIndustries/lorawan-stack-docs/commit/221d181de7bfac8dee2c0d9f61f49be595a166c5#diff-5ade1e9814fd2258512c15aa51eed853ec0c7d4e3f791aa507eea381917ac39a

closed 2021-07-12 by benolayinka