TheThingsIndustries/lorawan-stack-docs #222: Document difference between claiming and creating things in AWS IoT integration

johanstokking, 2021-02-08:
#### Summary

Document the difference between creating things and claiming things in the AWS IoT integration.

#### Why do we need this ?

For many users, it may not be clear what the difference is and when to use what.

#### What is already there? What do you see now?

Documentation here: https://www.thethingsindustries.com/docs/integrations/aws-iot/default/things/

#### What is missing? What do you want to see?

Begin with explaining what the difference is.

#### How do you propose to document this?

Introduce the differences.

It would be great if we can link to existing documentation about the concept of claiming end devices and manually creating end devices.

benolayinka, 2021-02-18:
@johanstokking I have no idea how claiming works - can you summarize the differences here, and I'll add to the docs?

johanstokking, 2021-02-18:
Claiming is documented here: https://www.thethingsindustries.com/docs/devices/device-claiming/

This issue is about shortly introducing the difference between creating devices and claiming (already existing) devices with a pointer to the URL above.

closed 2021-03-02 by benolayinka