TheThingsIndustries/lorawan-stack-docs #227: Break down command line instructions in to single lines

benolayinka, 2021-02-10:
#### Summary

To make instructions as simple as possible, let's break apart copy and paste command line instructions, like so: https://docs.mongodb.com/manual/tutorial/install-mongodb-on-ubuntu/#start-mongodb

benolayinka, 2021-02-25:
@nejraselimovic if you have time for this, please pick it up

nejraselimovic, 2021-03-01:
> @nejraselimovic if you have time for this, please pick it up

I can pick it up, but can you tell me an example from our docs? Everything I look at is way simpler when it's aggregated (otherwise there's too much re-typing), for example:

> $ ttn-lw-cli end-devices set $APP_ID $DEVICE_ID \
> --location.latitude 43.84 \
> --location.longitude 18.32 \
> --location.altitude 500 \

or can't really be broken into single lines, like this:

> $ ttn-lw-cli end-devices create app1 dev2 \
> --frequency-plan-id EU_863_870 \
> --lorawan-version 1.0.3 \
> --lorawan-phy-version 1.0.3-a \
> --abp \
> --session.dev-addr 00E4304D \
> --session.keys.app-s-key.key A0CAD5A30036DBE03096EB67CA975BAA \
> --session.keys.nwk-s-key.key B7F3E161BC9D4388E6C788A0C547F255

I guess it would make sense to do it *somewhere*, but it would help to know which content exactly you had in mind with this?

descartes, 2021-03-03:
FYI, I can select, copy & paste this format just fine - if anything, drop the $ to make it easier and put it in a block, maybe a block with a copy button like Mongo.

If it's all on one line it runs the risk of all sorts of confusion, at least this clearly shows each parameter.

Whilst I am do fight the good fight for the newbie, if you don't know this is spliting lines in the terminal, then perhaps you shouldn't be using the terminal.

PR #678 (2021-12-23): doc: Remove $ character from code blocks
#### Summary
Closes #227 and #200

#### Changes
- Removed $ character from command blocks to make it easier for people to copy commands
-
- 

#### Notes for Reviewers
WiP

#### Checklist
- [ ] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-ver [... cut]
closed 2021-12-23 by nejraselimovic