TheThingsIndustries/lorawan-stack-docs #232: How to backup/restore TheThingsStack

LasaleFamine, 2021-02-11:
#### Summary
I would love to see a guided doc to backup and restore The Things Stack of the Open Source installation.

#### Why do we need this ?
I'm currently looking to upgrade from `v3.10` to `v3.11` but I'm scared of the bad things that could happen so I want to backup the stack properly before upgrading the image and doing the migrations.

#### What is already there? What do you see now?
I couldn't find nothing similar to this.

#### What is missing? What do you want to see?
A dedicated page to backup/restore the stack maybe near the one for upgrading it.

#### How do you propose to document this?
Something like the upgrading guide: https://www.thethingsindustries.com/docs/getting-started/upgrading/

#### Can you do this yourself and submit a Pull Request?
I don't think so, unfortunately.

KrishnaIyer, 2021-02-11:
The Things Stack is designed to be stateless. The only backup you'll need to do are the databases and user uploaded content (ex: profile pictures).  That is outside the scope of TTS documentation.

LasaleFamine, 2021-02-12:
Thank you for you response @KrishnaIyer, but during the upgrade guide you need to make migration on the database itself, am I wrong? Also thinking about what you said, is it not worth to add a mention about this? I think the TTS docs it's pretty clear for every piece of the architecture and this seem to me a missing piece.

KrishnaIyer, 2021-02-15:
> you need to make migration on the database itself, am I wrong? Also thinking about what you said, is it not worth to add a mention about this?

You mean add a note to ask users to backup databases before migration? Sure. If you can make a PR that would be awesome.
Please check https://github.com/TheThingsIndustries/lorawan-stack-docs/blob/master/CONTRIBUTING.md for how to get started. If you cannot, then @benolayinka or @nejraselimovic can help.

descartes, 2021-03-03:
> The Things Stack is designed to be stateless. The only backup you'll need to do are the databases and user uploaded content (ex: profile pictures). That is outside the scope of TTS documentation.

This isn't helpful - it's not obvious that it's designed to be stateless and there could be something that's appropriate before backing up. And we don't all use the databases you have used, so some insight or link to resources wouldn't go amiss.

Also, where is the user uploaded content stored?

LasaleFamine, 2021-03-03:
Sorry for the late response @KrishnaIyer, but I'm not sure what need to be added. 
I'm asking here also because I don't know what I should take care of when updating the stack from the docker-compose file, apart the currently documented upgrade guide of course. 

I think that could be nice to have a piece of documentation taking care of backup and restore of the stack, since there is some data stored there, somewhere.

benolayinka, 2021-03-03:
@rvolosatovs can you chime in here with recommended ops for backing up TTS databases?

KrishnaIyer, 2021-03-04:
> @rvolosatovs can you chime in here with recommended ops for backing up TTS databases?

There really isn't any recommended ops since backups are dependent on the database used. We provide a docker compose file with Redis Server and Cockroach/Postgres containers as examples. Users are free to choose any Redis and Postgres compliant databases. 

This discussion is mixing 2 things.
1. **What** to backup during updates: @LasaleFamine is right here.  Indeed it's good to add a note on **what** needs to be backed up in the [`Upgrading`](https://www.thethingsindustries.com/docs/getting-started/upgrading/) section.
2. **How** to backup: This is out of scope. It's dependent on how the stack is deployed and the procedure varies slightly based on whether you run the databases in docker or run them as services etc. Also if you use another (managed) database, you'll need to check the docs of that.

In addition, I think that just listing what databases are used is sufficient. The users can then look up those databases. With a first pass on google, I found the following links.
Redis: https://redis.io/topics/persistence
Cockroach: https://www.cockroachlabs.com/docs/v21.1/take-full-and-incremental-backups.html
Postgres: https://www.postgresql.org/docs/\<version\>/backup.html

benolayinka, 2021-03-04:
> > @rvolosatovs can you chime in here with recommended ops for backing up TTS databases?
> 
> There really isn't any recommended ops since backups are dependent on the database used. We provide a docker compose file with Redis Server and Cockroach/Postgres containers as examples. Users are free to choose any Redis and Postgres compliant databases.
> 
> This discussion is mixing 2 things.
> 
> 1. **What** to backup during updates: @LasaleFamine is right here.  Indeed it's good to add a note on **what** needs to be backed up in the [`Upgrading`](https://www.thethingsindustries.com/docs/getting-started/upgrading/) section.
> 2. **How** to backup: This is out of scope. It's dependent on how the stack is deployed and the procedure varies slightly based on whether you run the databases in docker or run them as services etc. Also if you use another (managed) database, you'll need to check the docs of that.
> 
> We also need to update that note on DB migrations.
> 
> In addition, I think that just listing what databases are used is sufficient. The users can then look up those databases. With a first pass on google, I found the following links.
> Redis: https://redis.io/topics/persistence
> Cockroach: https://www.cockroachlabs.com/docs/v21.1/take-full-and-incremental-backups.html
> Postgres: [https://www.postgresql.org/docs/\](https://www.postgresql.org/docs/%5C)<version>/backup.html

Sounds good. In the upgrading guide, it looks like we currently mention only `is-db` migration. Is there anything else that needs to be backed up?

descartes, 2021-03-05:
> 2. **How** to backup: This is out of scope. It's dependent on how the stack is deployed and the procedure varies slightly based on whether you run the databases in docker or run them as services etc. Also if you use another (managed) database, you'll need to check the docs of that.

The instructions you provide to install the server are for **your** recipe. At the very least there should be some bullet points on backing up that configuration, particularly, as I said above, apparently there is some user uploaded content that's location is yet to be revealed.

If TTI want to make this a world-class offering, leaving out details on backups isn't really demonstrating good practise.

I don't think anyone who's capable of subverting the install instructions to change to a different database is going to have a problem figuring out how to backup their own databases.

One alternative is to document installing the stack without docker and allowing the use of alternative databases ...

KrishnaIyer, 2021-03-25:
@LasaleFamine: Sorry for the long wait. This PR should answer your questions; https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/285.

LasaleFamine, 2021-03-26:
Wow thanks @KrishnaIyer, this is actually even more than I expected. Great work!

PR #285 (2021-03-31): Add backup notes

#### Summary

Closes #232 

#### Changes

- Clean up `Upgrading The Things Stack` page.
- Add notes on backup and migrations.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONT [... cut]
closed 2021-03-31 by KrishnaIyer