TheThingsIndustries/lorawan-stack-docs #234: Document all of the json users see in the console events

benolayinka, 2021-02-16:
#### Summary

Let's
- Very clearly link to the [data formats](https://www.thethingsindustries.com/docs/reference/data-formats/) documentation in the console and somewhere in the getting started docs
- Add missing json information to that documentation. For example, what are the gateway metrics? What metrics do UDP gateways provide vs LBS?

#### Why do we need this ?

It's not obvious what most of the data reported in event view means

[image]

#### Can you do this yourself and submit a Pull Request?

I can add the information to the docs. @kschiffer , what's the best way to serve this info to users in context?

kschiffer, 2021-02-23:
Let's add this information to the event details panel on the right. We can add a message saying `For more information on specific event message types, please see our <DocLink>Data Formats</DocLink> documentation`. This should be added below the json.

@benolayinka it would be nice if we could also link to individual message types.

benolayinka, 2021-02-25:
> Let's add this information to the event details panel on the right. We can add a message saying `For more information on specific event message types, please see our <DocLink>Data Formats</DocLink> [...]

Can you do `<DocLink>/reference/data-formats#join-accept-messages</DocLink> documentation`

asmulko, 2021-02-25:
> Can you do `<DocLink>/reference/data-formats#join-accept-messages</DocLink>`

@benolayinka you can take a look at screenshot how it will look now in this pr https://github.com/TheThingsNetwork/lorawan-stack/pull/3859. Do you want any changes?

kschiffer, 2021-02-25:
> Can you do `<DocLink>/reference/data-formats#join-accept-messages</DocLink>`

We could do that but I think we would need to restructure the docs a little for that. It should be possible to have a hashlink for every message type name, e.g. `Successfully processed data message` -> `/reference/data-formats#successfully-processed-data-message` or `Drop data message` -> `/reference/data-formats#drop-data-message`.

closed 2021-03-02 by asmulko