TheThingsIndustries/lorawan-stack-docs #240: Search feature is broken

rish-c, 2021-02-19:
#### Summary
Search function on the docs is broken (and has been for over a month now)

#### Why do we need this ?
Easier navigation 

#### What is already there? What do you see now?
The search function used to work but I think since 3.10 it is broken

#### What is missing? What do you want to see?
The search field is there but it doesn't work properly. 
Maybe if this 3rd party search add on isn't good, we can look for an alternate option.

#### Can you do this yourself and submit a Pull Request?
@benolayinka @nejraselimovic

rish-c, 2021-02-19:
Can someone assign and add the labels etc. I dont have rights to do so.

benolayinka, 2021-02-22:
I've been emailing them for a month. Will keep trying, Algolia is the best and most popular option

closed 2021-02-23 by benolayinka