TheThingsIndustries/lorawan-stack-docs #244: [bug] Clicking an entry in Search results does not open the corresponding documentation page

lnlp, 2021-02-25:
_This is not a documentation request for missing contents but reporting of an issue with the documentation site.
There currently is no option to select for that when creating a new issue._

#### Summary
Search function on the documentation site does not work correctly.
Clicking an entry in Search results does not open the corresponding documentation page.

#### Why do we need this ?
Search should work correctly and as expected.
If not this results in poor UX causing frustration.

#### What is already there? What do you see now?
Searching for 'payload' in order to get to the Payload Formatting documentation shows the following search results:

[image]

"Payload Formatters" that is exactly what I was looking for.
Clicking 'Payload Formatters' should forward me to 
https://www.thethingsindustries.com/docs/integrations/payload-formatters/

It does not however and instead forwards to https://thethingsindustries.com/docs/integrations/
which is not the Payload Formatters page (but 'one level higher').

This incorrect behavior is consistent, independent of the search term used.

#### What is missing? What do you want to see?
Clicking on an entry in the Search results should open the correct corresponding documentation page.

#### How do you propose to document this?
_Not relevant._

#### Can you do this yourself and submit a Pull Request?
No.

descartes, 2021-03-03:
The search rarely returns obvious results - I've tested this by searching for keywords in the docs and get back no results.

benolayinka, 2021-03-08:
This should be fixed now. It was caused by the migration from thethingstack.io to thethingsindustries.com/docs.

closed 2021-03-08 by benolayinka