TheThingsIndustries/lorawan-stack-docs #262: Document guide for Tektelic Kona Micro Lite IoT Gateway to connect to TTS

i3mSuperMan, 2021-03-10:
#### Summary
There is no documentation guide for the `Tektelic Kona Micro Lite` Gateway to connect to The Things Stack.
#### Why do we need this?
To connect Kona Micro Lite Gateway to The Things Stack.
#### What is already there? What do you see now?
We see a configuration guide for the `Tektelic Kona Micro` IoT LoRaWAN Gateway
Ref: https://www.thethingsindustries.com/docs/gateways/tektelickonamicro/
#### What is missing? What do you want to see?
A configuration guide for the `Tektelic Konal Micro Lite` Gateway to connect to TTS.
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No. @benolayinka

laurensslats, 2021-04-29:
Moving to next up as the configuration steps are similar to the Tektelic Kona Micro

nejraselimovic, 2021-06-15:
I'm closing this issue as it has been already addressed but PR probably wasn't linked to this issue: https://www.thethingsindustries.com/docs/gateways/tektelickonamicro/.

closed 2021-06-15 by nejraselimovic