TheThingsIndustries/lorawan-stack-docs #266: Move LoRa Cloud documentation and add Console

johanstokking, 2021-03-18:
#### Summary

Make the LoRa Cloud integration easier to find and don't only mention CLI steps

#### Why do we need this ?

Better navigation and less CLI

#### What is already there? What do you see now?

https://www.thethingsindustries.com/docs/reference/application-packages/lora-cloud-device-and-application-services/

#### What is missing? What do you want to see?

It shouldn't be buried under Application Packages; users don't care about the technical difference between packages and webhooks for example.

Also it only contains CLI steps, but we have Console support now.

#### How do you propose to document this?

1. Move it under Cloud Integrations, or maybe on the root level in Integrations? What is the logic from a user's perspective to find things under Integrations or under Cloud Integrations? Isn't AWS IoT also a cloud integration?

2. Add the Console instructions, probably with tabs like we do elsewhere now

#### Can you do this yourself and submit a Pull Request?

Loop in @adriansmares for review

johanstokking, 2021-06-09:
Please aim for closing this in Q2.

nejraselimovic, 2021-06-09:
> Please aim for closing this in Q2.

I already started working on this

johanstokking, 2021-06-14:
This also needs some documentation about how things work, especially Geolocation.

1. What is TDOA? What is multi-frame? Is automatic multi-frame always enabled?
2. What is GNSS?
3. What is the payload format of GNSS? (we don't know, it's opaque to TTS)
4. Is all payload sent to the Geolocation service? (afaik only on the default port?)

Looping in @adriansmares to help answering these questions.

PR #407 (2021-06-30): Improve LoRa Cloud integration docs
#### Summary
Closes #266 

#### Changes
Moving LoRa Cloud docs from Reference -> Application packages to Integrations -> LoRa Cloud, adding information about DAS and Geolocation integrations in separate sections, adding Console steps

#### Notes for Reviewers
There are two things that are on to-do list and I didn't do yet, but didn't want to drag with this PR anymore:
- I didn't include CLI docs in the PR (they are not ready yet). Having Console steps properly documented is probably way more important
- I will blur the sensitive data (locations, tokens..) an [... cut]
closed 2021-06-30 by nejraselimovic