TheThingsIndustries/lorawan-stack-docs #271: Improve CLI doc UI

benolayinka, 2021-03-22:
#### Summary

A user built his own CLI docs and recommended we make our own CLI doc navigation more similar for ease of use -> http://descartes.co.uk/TTSv3CLI.html

descartes, 2021-06-21:
> A user

Excuse me, I'm no user, I'm a contributor ;-)

benolayinka, 2021-06-28:
> > A user
> 
> Excuse me, I'm no user, I'm a contributor ;-)

Haha yes of course! Is the source for your docs public?

descartes, 2021-06-28:
> Haha yes of course! Is the source for your docs public?

Yes, it's all one web page - and very simple stuff

benolayinka, 2021-06-28:
> > Haha yes of course! Is the source for your docs public?
> 
> Yes, it's all one web page - and very simple stuff

How are you scraping and generating your docs?

descartes, 2021-07-01:
Answered on the other thread!

benolayinka, 2021-08-30:
blocked on https://github.com/TheThingsNetwork/lorawan-stack/issues/4378

benolayinka, 2021-10-18:
@descartes this is finally done :) #594