TheThingsIndustries/lorawan-stack-docs #274: Documentation on how the ADR algorithm works in TTS V3

i3mSuperMan, 2021-03-22:
#### Summary
To provide information to the users on how exactly the ADR algorithm works in TTS V3.
#### Why do we need this?
This will make the users understand how `ADR` was implemented and the behavior in different frequency bands.
#### What is already there? What do you see now?
Currently, there is no documentation with the ADR details.
#### What is missing? What do you want to see?
Documentation on

- The `ADR` concept
    - How does the ADR algorithm work in TTS V3?
    - What parameter values (RSSI/SNR/....) does it take into consideration?
- Which parameter (DataRate/TX_Power/NBTrans) is dependent on which parameters (RSSI/SNT/....)?
- The behavior in the different frequency bands
- Customer ADR and related commands

#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No.

benolayinka, 2021-06-07:
@johanstokking @htdvisser where is a good place for me to figure out how ADR works in TTS?

htdvisser, 2021-06-08:
We may have some upcoming changes to the ADR algorithm (https://github.com/TheThingsNetwork/lorawan-stack/issues/4092) but the general idea will stay the same.

Our ADR implementation looks at the SNR to determine the Data Rate and Tx Power, and looks at frame counters to determine packet loss, and set the number of re-transmissions accordingly.

The implementation is based on Semtech's recommended algorithm described in [this document](https://www.thethingsnetwork.org/forum/uploads/default/original/2X/7/7480e044aa93a54a910dab8ef0adfb5f515d14a1.pdf):

- We determine [the maximum SNR over recent transmissions](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.13/pkg/networkserver/mac/adr.go#L218)
- We determine [the minimum SNR we'd need to demodulate an uplink given the current parameters](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.13/pkg/networkserver/mac/adr.go#L232)
- We calculate [the margin we have to further optimize the data rate](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.13/pkg/networkserver/mac/adr.go#L236)
	- Part of this is configurable per device (if you use the CLI)
	- If we have less measurements (uplinks) than we want, [we include a safety margin](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.13/pkg/networkserver/mac/adr.go#L238-L240)
- We [increase the data rate as long as there's enough margin](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.13/pkg/networkserver/mac/adr.go#L251-L262)
- If there's still margin after reaching the maximum data rate, [we decrease the transmit power](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.13/pkg/networkserver/mac/adr.go#L273-L281)
- Depending on packet loss, [we increase the number of retransmissions](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.13/pkg/networkserver/mac/adr.go#L288-L296)

closed 2021-06-25 by benolayinka