TheThingsIndustries/lorawan-stack-docs #318: Enable Packet Broker configuration via CLI

laurensslats, 2021-04-22:
#### Summary
Packet Broker commands are failing, possible configuration step for TTS Clous tenants is missing

#### Why do we need this ?
Allow customers to see their PB configuration, and make changes  

#### What is already there? What do you see now?
Documentation available: https://www.thethingsindustries.com/docs/reference/packet-broker/configure/

#### What is missing? What do you want to see?
Issues:
`$ ttn-lw-cli packetbroker registration info`
returns the overview of packetbroker commands:
[image]

`$ ttn-lw-cli packetbroker register`
return:
```
WARN [core]grpc: addrConn.createTransport failed to connect to {localhost:8884 localhost:8884 <nil> 0 <nil>}. Err: connection error: desc = "transport: error while dialing: dial tcp [::1]:8884: connect: connection refused". Reconnecting...
connection error: desc = "transport: error while dialing: dial tcp [::1]:8884: connect: connection refused"
```

Supposing that some PB configuration is required (`ttn-lw-stack.yaml` file?), but documentation to do so for TTS Cloud is missing.

Only documentation for configuring PB for `Enterprise` & `Open Source` are available:
https://www.thethingsindustries.com/docs/reference/packet-broker/connect/

#### How do you propose to document this?
Request input from @johanstokking 
Documentation can be updated by myself

i3mSuperMan, 2021-04-22:
Here are a few more points:

1. We see that `packet-broker-agent-grpc-address` has to be set to access the PB with CLI. This address is automatically included in the config file generate via `ttn-lw-cli use thethings.example.com` command. However, we think it has to be documented in the [Configure Packet Broker](https://www.thethingsindustries.com/docs/reference/packet-broker/configure/) and also in the [manually create the configuration file](https://www.thethingsindustries.com/docs/getting-started/cli/installing-cli/#manually-create-configuration-file) reference.
2. Also, it’s worth adding the below command to the [routing policies](https://www.thethingsindustries.com/docs/reference/packet-broker/configure/routing-policies/) doc that sets routing policy from TTSC tenant to `ttn` as it is a common use case.
```
ttn-lw-cli packetbroker home-network policy set 000013 ttn --all
```

johanstokking, 2021-04-22:
Fixed in #319.

Please run `ttn-lw-cli use tti.eu1.cloud.thethings.industries` to update your configuration file.

This is one of the issues that occur with the CLI today: each (new) component needs an explicit address in the configuration file. I scheduled https://github.com/TheThingsNetwork/lorawan-stack/issues/1230 for this quarter to fix this.

The API key that TTSC was using wasn't having the correct rights, but that's fixed now too, but it takes up to an hour for the security token to refresh and for this to become effective.

PR #319 (2021-04-28): Fix and clarify Packet Broker documentation

#### Summary

Closes #318

#### Screenshots

When `hint-upgrade` is set:

[image]

#### Changes

- Add `hint-upgrade` param to `cli-only`
- Fix `ttn-lw-cli packetbroker registration` command
- Small clarifications

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [x] New Features Marked: Documentation  [... cut]
closed 2021-04-28 by johanstokking