TheThingsIndustries/lorawan-stack-docs #325: Using distribution shortcode with a subsection title makes it disappear from right-hand navigation bar

nejraselimovic, 2021-04-28:
#### Summary
Storage integration subsection with marked distributions missing in the right-hand navigation:
[image]

Storage integration subsection appears in the right-hand navigation when distributions shortcode gets removed:
[image]

benolayinka, 2021-06-07:
Having shortcodes inline with headings is causing us quite a few problems. It messes up anchor links, and needs some manual fixes to work. In Hugo documentation, they move the shortcodes to the text below the heading, which would look like this for us:

[image]

This is a really easy fix and my preference. The other option is to wrap entire sections in some kind of background or indicator, like

[image]

nejraselimovic, 2021-06-07:
The first option seems better indeed

PR #382 (2021-06-10): Move inline shortcodes to next line

#### Summary

Shortcodes in headings were breaking anchor links and causing headaches, so moving them to the first line of the next paragraph, like HUGO documentation does -> https://gohugo.io/variables/site/#get-the-site-object-from-a-partial

Closes #325 

#### Screenshots

[image]

[image]

[image]

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if st [... cut]
closed 2021-06-10 by johanstokking