TheThingsIndustries/lorawan-stack-docs #330: Include regional instances information of Packet Broker

i3mSuperMan, 2021-04-29:
#### Summary
Update the Packet Broker documentation to include the latest packet broker deployments and their routing policies.

In the latest updates to PB clusters, the regional instances of TTN v2, TTSC, and TTN V3 are wired with the corresponding regional instances of Packet Broker.
Ref: https://github.com/TheThingsIndustries/lorawan-stack-support/issues/406#issuecomment-824921495

#### Why do we need this ?
To make it clear to the users about which cluster of TTSC V3 is wired to which regional packet broker instance and the data routing restrictions.

For example, packet broker will not route data from a gateway connected to the `au` cluster of TTN V3 or an AU915 gateway in TTN V2 public to the devices present in the `eu1` cluster of TTSC V3 and vice versa.

#### What is already there? What do you see now?
Information related to packet broker and how to configure PB.
Ref: https://www.thethingsindustries.com/docs/reference/packet-broker/

#### What is missing? What do you want to see?
Information about the latest addition in the packet broker and the latest changes to the routing functionality.

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?
No

benolayinka, 2021-05-05:
From https://github.com/TheThingsIndustries/lorawan-stack-support/issues/406#issuecomment-824921495 it looks like PB was rewired so that `au1`, `eu1` and `nam1` clusters of TTN V2, TTN V3, and TTSC all only route to the same geographical area. Is that right? @KrishnaIyer ?

KrishnaIyer, 2021-05-05:
Yes but cross region routing is being worked on.

closed 2021-05-06 by benolayinka