TheThingsIndustries/lorawan-stack-docs #357: V3 Storage Integration - New Retrieve parameter, Last X Minutes

sierrasprout, 2021-05-19:
#### Summary

Regarding the V3 storage integration API, would you consider adding a “last X minutes” option to the list of fields in the Retrieve messages HTTP API? 
...

#### Why do we need this ?

This would be a helpful alternative to the --after parameter and ease our migration (and others) from V2.
...

#### What is already there? What do you see now?

https://www.thethingsindustries.com/docs/reference/api/storage_integration/

...

#### What is missing? What do you want to see?

Add an additional field for Message Retrieval, similar to "after", but indicates the last X minutes (or other unit of time).
...

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?

Unfortunately I don't have the expertise to do this myself.
...

nejraselimovic, 2021-05-21:
There is a duplicate of this issue in another repository: https://github.com/TheThingsNetwork/lorawan-stack/issues/4198, and I see an ongoing discussion there, so I will temporarily close this issue. If there's a need, I will reopen it. Please take a look at that issue for further info.

closed 2021-05-21 by nejraselimovic

sierrasprout, 2021-05-21:
Thank you - I'll check it out.