TheThingsIndustries/lorawan-stack-docs #358: Update the `JSON File Reference` in `Migrating to The Things Stack` documentation

i3mSuperMan, 2021-05-20:
#### Summary
The  **JSON File Reference** section in the **Migrating to The Things Stack** documentation is not up to date concerning the `3.12.0` API changes.

#### Why do we need this ?
If users try to add their devices by following the JSON File Reference, they will encounter issues.

#### What is already there? What do you see now?
In the current documentation, the fields are based on the older version of APIs.
For example, the field for Rx1Delay is mentioned as `mac_settings.rx1_delay.value`.  

#### What is missing? What do you want to see?
Updating the documentation with the latest API changes in the `TTS v3.12.0` version.
I.e., the field for **Rx1Delay** should be `mac_settings.rx1_delay` as per the latest API changes.

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?
No. @benolayinka

benolayinka, 2021-05-25:
@neoaggelos @laurensslats do you know where to find these API values?

neoaggelos, 2021-05-25:
I can pick this up.

PR #389 (2021-06-22): Update Rx1 delay field name according to 3.12.0 API

#### Summary

Closes #358 

#### Checklist

- [ ] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [ ] Style Guidelines: Documentation obeys style  [... cut]
closed 2021-06-22 by neoaggelos