TheThingsIndustries/lorawan-stack-docs #371: Sub-section delimiters to make documentation easier to track

descartes, 2021-05-30:
#### Summary
The design is nice but ruled lines between the details of a command or message but not bolder & blacker line between actual items makes finding information a blur

#### Why do we need this ?
Some of our eye balls are 50+ years old, I use three pairs of glasses now and if I'm scrolling up & down I often miss the detail I'm looking for. And as most of the migration work at present is not revenue generating, my time is a little precious to have to fight with the design.

#### What is already there? What do you see now?
[image]

[image]

#### What is missing? What do you want to see?
[image]

[image]

#### Can you do this yourself and submit a Pull Request?
Possibly - if only as I find Hugo interesting for my own use and also so I can recycle the content in to something as horrible as this: http://descartes.co.uk/TTSv3CLI.html which I will probably do anyway as a drill down allows people to see the big picture and then get in to the detail they need.

descartes, 2021-05-30:
Actually, now I look at it, please consider dropping the line between the details - and perhaps right align the left hand column.

benolayinka, 2021-06-08:
Fixed in #385 - take a look @descartes and let me know if it looks okay

PR #385 (2021-06-09): Update API doc styling

#### Summary

Closes #371 

#### Screenshots

[image]

#### Changes

- Add border between tables
- Remove borders inside tables
- Right align headers

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the gui [... cut]
closed 2021-06-08 by descartes