TheThingsIndustries/lorawan-stack-docs #379: Document adding 16 channel Gateway in the TTS

i3mSuperMan, 2021-06-07:
#### Summary
The Things Stack already supports 16 channel UDP and LBS( [Concentrator Design v1.5](https://doc.sm.tc/station/gw_v1.5.html)) Gateways. To configure a gateway with an additional frequency plan, Users need to set the `--frequency-plans-ids` option which accepts an array, and note that both frequency plans must be from the same PHY region.
`ttn-lw-cli gateways update <Gateway ID> --frequency-plan-ids EU_868_870 //or whichever`
#### Why do we need this?
To connect `16 Channel` UDP and LBS Gateways with TTS.
#### What is already there? What do you see now?
Adding `8 Channel` Gateways to TTS. Ref:https://www.thethingsindustries.com/docs/gateways/adding-gateways/
#### What is missing? What do you want to see?
Adding `16 Channel` Gateways to TTS.
#### How do you propose to document this?
Ref: https://github.com/TheThingsIndustries/lorawan-stack-support/issues/413#issuecomment-830016601
#### Can you do this yourself and submit a Pull Request?
No. @benolayinka

KrishnaIyer, 2021-07-23:
Nope I don't have a gateway.

KrishnaIyer, 2021-09-16:
We don't have a gateway and the vendor we asked for assistance did not respond. Removing the milestone.