TheThingsIndustries/lorawan-stack-docs #400: Documentation on the TTN Mapper integration with TTS

i3mSuperMan, 2021-06-15:
#### Summary
We see that there is `TTN Mapper webhook` integration for the applications in the `TTS console` and this issue to request related documentation.
[Image]
#### Why do we need this?
To successfully integrate TTS applications with the TTN Mapper
Ref: https://ttnmapper.org/
#### What is already there? What do you see now?
https://www.thethingsindustries.com/docs/integrations
#### What is missing? What do you want to see?
Documentation on TTN Mapper integration with TTS
#### How do you propose to document this?
Add documentation according to the Github `ttnmapper/documentation`. 
Ref1: https://github.com/ttnmapper/documentation/blob/master/integration/tts-integration-v3.md
Ref2: https://github.com/TheThingsNetwork/lorawan-webhook-templates/pull/21#issuecomment-802698922
#### Can you do this yourself and submit a Pull Request?
No. @benolayinka

nejraselimovic, 2021-06-15:
@i3mSuperMan thanks, I'm picking this up soon

benolayinka, 2021-06-18:
> @i3mSuperMan thanks, I'm picking this up soon

I don't think you really need to add that documentation to The Things Stack - it's TTN mapper specific, not TTS specific. You could just add a link to that documentation in the template somewhere.

If you're excited to do it anyway, I guess it's good content to have