TheThingsIndustries/lorawan-stack-docs #406: Whitelisting Server IP addresses information in the Networking docs

i3mSuperMan, 2021-06-16:
#### Summary
The Things Stack uses a port per the protocol and we have default ports listed [here](https://www.thethingsindustries.com/docs/reference/networking/#port-allocations) that should be allowed to connect with TTS. Information on whitelisting Server IP addresses would be needed for few customer scenarios.
#### Why do we need this?
Using The Things Stack behind the firewall.
#### What is already there? What do you see now?
The Current documentation contains the `Default Ports` used by the protocols in TTS. Ref: https://www.thethingsindustries.com/docs/reference/networking/#port-allocations
#### What is missing? What do you want to see?
Information on the Server IP addresses, i.e., whitelisting IP addresses information while using TTS.

**Insights:**
**The Things Stack Cloud** clusters are currently deployed to the following AWS regions:
```
eu1: eu-west-1
nam1: us-west-1
au1: ap-southeast-2
```
Ref:[ The Things Stack Cloud](https://www.thethingsindustries.com/docs/getting-started/cloud-hosted/)
- As we are working with the public cloud, it is possible that the IP addresses change and we can’t make any commitments to keeping the same IP addresses over time. One can refer to the [AWS IP address ranges](https://docs.aws.amazon.com/general/latest/gr/aws-ip-ranges.html) to know the IP addresses range of the above-mentioned AWS regions.
- To use dedicated IP addresses, users might opt to have a dedicated cloud-hosted deployment. Ref: [Deployment Options](https://www.thethingsindustries.com/deployment/)
- If hosting on your own, as per the [ECS docs](https://www.thethingsindustries.com/docs/getting-started/aws/ecs/architecture/#vpc-and-subnets),

> All incoming connections to The Things Stack will use the IP addresses of the NLBs, and all outgoing connections will use the IP addresses of the NGWs. **This means that you can use these IP addresses in the firewall rules for gateways that need to connect to your deployment**, and on other servers that need to connect to your deployment or receive connections (webhooks) from it.

#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No. @benolayinka

benolayinka, 2021-06-16:
I'm not sure what you mean - if using The Things Stack Cloud, TTI manages the instance and port allocation. You just need to be able to use HTTPS to access the console. Or do you mean that some firewalls are blocking access to the ECS IP address that TTS Cloud is hosted on? Can't they just whitelist the DNS address instead?

ymgupta, 2021-07-15:
@benolayinka 

As you rightly pointed, suggested method would be to whitelist the DNS. But few users of The Things Stack Cloud request if they can whitelist IP addresses instead of DNS address. So, we need to document the recommended way is to whitelist DNS and IP addresses change as we work with public cloud. However, they can refer the AWS reference to know and whitelist the entire address range if they prefer that way.

closed 2021-07-28 by benolayinka