TheThingsIndustries/lorawan-stack-docs #409: Document differences in TTI stack level implementation of LBS protocol with reference to LBS standard.

i3mSuperMan, 2021-06-17:
#### Summary
As per https://github.com/TheThingsIndustries/lorawan-stack-support/issues/451, TTS implements gateway scheduler different from LBS scheduler for good reasons. It is better to document it.
#### Why do we need this?
Document the differences and make sure gateway manufacturers know about these differences.
#### What is already there? What do you see now?
...
#### What is missing? What do you want to see?
Differences in TTI stack level implementation of LBS protocol with reference to LBS standard
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No. @benolayinka 

cc: @KrishnaIyer

benolayinka, 2021-06-18:
i read the thread and my best understanding is that TTS modifies `xtime` to account for `RX1Delay`. Is that right? Is that all I need to document? What information could that provide for gateway manufacturers (maybe that will help me figure out where to put it in the docs)

@KrishnaIyer

KrishnaIyer, 2021-07-12:
I'll take this one.

PR #614 (2021-10-08): Document LBS downlink scheduling

#### Summary

 
Closes #409 

#### Changes

- Document LBS downlink scheduling

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [x] Style Guide [... cut]
closed 2021-10-08 by KrishnaIyer