TheThingsIndustries/lorawan-stack-docs #414: CLI end device options lacking clarity

descartes, 2021-06-21:
#### Summary
There are command line options for current, desired, proposed, pending and other some such.

I figure the current should be read only, the pending is what will be applied on next contact with the device but this is an educated guess and I'm not totally clear on which fields to set.

#### Why do we need this ?
Save confusion on which fields are read-only & which are write-only & which are read-write

#### What is already there? What do you see now?
Nothing that speaks to this issue.

#### What is missing? What do you want to see?
Some tags to show RO, WO & RW in the the docs & in the CLI

#### How do you propose to document this?
Sadly, not me, as I don't know the official answers

#### Can you do this yourself and submit a Pull Request?
Sorry, nope.

benolayinka, 2021-06-21:
@descartes , will follow up on this as part of our CLI docs overhaul. I'd also love to see the source for http://descartes.co.uk/TTSv3CLI.html to borrow the layout if you don't mind!

descartes, 2021-06-28:
> I'd also love to see the source for http://descartes.co.uk/TTSv3CLI.html to borrow the layout if you don't mind!

Right click on the browser page! It's a perfect example of a quick hack knocked up one evening out of frustration.

benolayinka, 2021-06-29:
> > I'd also love to see the source for http://descartes.co.uk/TTSv3CLI.html to borrow the layout if you don't mind!
> 
> Right click on the browser page! It's a perfect example of a quick hack knocked up one evening out of frustration.

Sorry I mean I'm curious about how you're generating it, especially how you're generating/scraping the command tree.

descartes, 2021-07-01:
Ah, OK, I have a script that recursively issues help commands, captures the output and looks for sub-commands. Then other parsing scripts go through cutting up the output in to different sections. Then another script pieces the pieces back together to create one huge web page (so no latency, otherwise I'd use XHR).

descartes, 2021-07-01:
In the meanwhile, as I'm fixing things for people on the forum with CLI command suggestions, could someone please answer the original query in brief so I can crack on whilst you update the docs

johanstokking, 2021-07-01:
Here you go: https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.13/pkg/ttnpb/field_mask_validation.go

Besides this, the `ttn-lw-cli end-device {get,set} --help` commands shows the flags of fields that can be read and written respectively.

closed 2021-10-19 by benolayinka