TheThingsIndustries/lorawan-stack-docs #430: Document a guide for AWS SaaS deployment configuration.

i3mSuperMan, 2021-07-03:
#### Summary
Document the instructions for deploying The Things Stack as AWS SaaS deployment along with the configuration.
Ref: https://github.com/TheThingsIndustries/lorawan-stack-support/issues/364

#### Why do we need this ?
To help users set up TTS as AWS SaaS deployment and to troubleshoot the issues.

#### What is already there? What do you see now?
We have TTS available as AWS SaaS deployment

#### What is missing? What do you want to see?
The documentation on deploying The Things Stack as AWS SaaS deployment.

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?
No. @benolayinka

benolayinka, 2021-07-12:
I don't understand what needs to be updated here. @adriansmares ?

i3mSuperMan, 2021-08-02:
Hi @adriansmares,

A gentle reminder on this.

NicolasMrad, 2022-03-29:
what's the status here?

michalborkowski96, 2022-03-30:
@NicolasMrad I don't really know much about using SaaS, probably @adriansmares would be a better match for this issue

adriansmares, 2022-03-31:
I will pick this up in Q2.

ymgupta, 2023-02-22:
Below are the steps and example screenshots for subscribing to the tenant in The Things Stack Cloud deployment via AWS Marketplace:

**Step 1:** Navigate to [The Things Stack Cloud for LoRaWAN](https://aws.amazon.com/marketplace/pp/prodview-djx65iyt6w5cy) in AWS Marketplace. 

[image]

**Step 2:** Log in to the AWS account and click on `View purchase options`. Then we will be navigated to the below page and click on `Subscribe`.

[image]

**Step 3:** Navigate to the Manage Subscriptions in the AWS Marketplace.

[image]

**Step 4:** In The Things Stack Cloud for LoRaWAN subscription, select the option `Set up product` at the bottom.

[image]

**Step 5:** You will observe a pop up, i.e., `To continue, set up your account and complete the registration`. Then Click on `Set up your account`.

[image]

**Step 6:** Fill in all the details on the `Configure your account` page and click on submit.

[image]

After completion of the above steps, users can get the tenant ID and the user details to access The Things Stack.

@nejraselimovic, could you please document the above process to subscribe a tenant via AWS Marketplace in The Things Stack Cloud deployment?

cc: @adriansmares

KrishnaIyer, 2024-02-29:
@ymgupta: Since you already have all the information and screenshots, why not just open a PR yourself?

Siva-7482, 2024-04-08:
@KrishnaIyer, we have filed a PR with the above mentioned information and screenshots. Please review.

Ref: https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/1305

ymgupta, 2024-06-15:
This is completed in the below PR.
Ref: https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/1305

closed 2024-06-15 by ymgupta