TheThingsIndustries/lorawan-stack-docs #431: Add configuration section to AWS AMI guide

i3mSuperMan, 2021-07-03:
#### Summary
There is no information in the [AWS Marketplace AMI](https://www.thethingsindustries.com/docs/getting-started/aws/ami/) guide on how to configure The Things Stack.

#### Why do we need this?
To let users know how to configure The Things Stack deployed as AWS AMI.

#### What is already there? What do you see now?
We have instructions to deploy TTS as AWS AMI and the troubleshooting guide.
Ref: https://www.thethingsindustries.com/docs/getting-started/aws/ami/

#### What is missing? What do you want to see?
Information on how to configure The Things Stack.

#### How do you propose to document this?
We have the dedicated [Configuration](https://www.thethingsindustries.com/docs/reference/configuration/) guide to configure TTS. So, add a section about configuring TTS in AWS AMI guide and provide the [Configuration](https://www.thethingsindustries.com/docs/reference/configuration/) guide reference.

#### Can you do this yourself and submit a Pull Request?
@benolayinka

nejraselimovic, 2021-07-07:
I think this needs help from @johanstokking or @htdvisser as well

benolayinka, 2021-07-07:
Also, what do you mean by configure? Can you give an example of a problem that came up that made you raise this issue?

benolayinka, 2021-07-15:
@i3mSuperMan ?

ymgupta, 2021-07-15:
@benolayinka Sorry for the delay in getting back on this.

This is about letting AWS AMI Users know about [Configuration](https://www.thethingsindustries.com/docs/reference/configuration/) that is possible/available for them. In the guides of other deployments (ECS or Open Source), we have stated as below:

> For the other parameters, see the descriptions and the [Configuration](https://www.thethingsindustries.com/docs/reference/configuration/) reference.

It is good to have the same mentioned in AWS AMI [Post deployment guide](https://www.thethingsindustries.com/docs/getting-started/aws/ami/post-deployment/).

johanstokking, 2021-07-16:
Those are really not applicable for the AWS AMI users, besides the CloudFormation parameters that we provide as part of the template.

So we need to do two things to close this:
1. We need to state on that **Configuration** page that it applies to TTSOS and TTSE only, and maybe even clarify further that it does not apply to the AWS Launcher
2. We need to know which things need to be configured for the AWS Launcher. We add those as CloudFormation parameters

The thing is that we do not support manual editing of the configuration file of TTS. It is technically possible, but that will lead to problems when the machine gets updated of even restarted, as the CFT settings are leading.

i3mSuperMan, 2021-08-07:
Hi @benolayinka,

A gentle reminder on this. Please take this forward.

closed 2021-08-19 by benolayinka