TheThingsIndustries/lorawan-stack-docs #443: Mark fields which are only returned by field mask in API

benolayinka, 2021-07-08:
#### Summary

The ids , created_at, updated_at and deleted_at are always returned by the IS. Everything else needs to be selected.

#### Why do we need this ?

So people aren't confused when fields aren't returned by API

#### Can you do this yourself and submit a Pull Request?

Yes

NicolasMrad, 2022-03-29:
Is this still relevant or to be done soon since it has been moved for 3 quarters so far

benolayinka, 2022-04-04:
This is partially addressed by https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/475

Actually adding a field mask indicator will require manually updating request metadata which we aren't gonna do right now. Will mark gold plating and remove milestone

https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/475#issuecomment-884096007