TheThingsIndustries/lorawan-stack-docs #479: Separate page for disabling storage integration

i3mSuperMan, 2021-07-24:
#### Summary

Currently, in the storage integration documentation, we have the `Enable` page which also has steps to `disable` the storage integration. It would be good to have a separate page to `disable` the storage integration. Also, the command to disable the storage integration for an end device.
Ref: https://www.thethingsindustries.com/docs/integrations/storage/enable/#disable-the-storage-integration

#### Why do we need this?

For better organizing the information about storage integration

#### What is already there? What do you see now?

The process to disable the storage integration for an application is present in the Enable storage integration page.

#### What is missing? What do you want to see?

Separate page for disabling the storage integration just like we have to enable it. And also, the command to disable the integration for an end device using CLI.

#### How do you propose to document this?

Move the disable storage integration information to a new page below the Enable storage integration. And, document the below commands to disable the storage integration for an end device

1. List default associations

```
$ ttn-lw-cli applications packages associations list "app1" "dev1"
{
  "associations": [
    {
      "ids": {
        "end_device_ids": {
          "device_id": "dev1",
          "application_ids": {
            "application_id": "app1"
          }
        },
        "f_port": 100
      },
      "created_at": "2020-08-24T21:09:44.649890166Z",
      "updated_at": "2020-08-24T21:09:44.649890166Z",
      "package_name": "storage-integration"
    }
  ]
}
```
2. Disable the storage integration.
```
$ ttn-lw-cli applications packages associations delete "app1" "dev1" 100
```
#### Can you do this yourself and submit a Pull Request?
No, @benolayinka

benolayinka, 2021-08-04:
Did a customer ask for this?

i3mSuperMan, 2021-08-05:
No, @benolayinka.

But, we see that the title of the page says `Enable`, whereas it also has information about `disabling` the storage integration. We think it would be better to move the information about disabling the storage integration to a new page along with the relevant commands.

closed 2021-08-17 by benolayinka