TheThingsIndustries/lorawan-stack-docs #480: Add a note for configuring the Factory Preset Frequencies in adding devices documentations.

i3mSuperMan, 2021-07-26:
#### Summary
An ABP device should transmit uplinks only in default frequencies until the Network Server communicates the other non-default channels via MAC Commands. However, most of the devices transmit uplinks in the non-default frequencies, resulting in the drop of these packets by the Network Server. It might result in an inconsistent device state following unexpected issues.

To avoid this, the user needs to configure the `Factory Preset Frequencies` so that the NS will accept uplinks in all the channels. It is valid only for non 64-channel frequency plans (**EU868**/**IN865**/**AS923**/...).

CC: @adriansmares to confirm and add any missing inputs.

#### Why do we need this ?
To let users be aware of the setting as most of the users are not aware of this behavior and in the assumption that there might be an issue at the NS end.

#### What is already there? What do you see now?
In the MAC Settings guide, it was mentioned how to configure Factory Preset Frequencies for ABP devices.
Ref: https://www.thethingsindustries.com/docs/devices/mac-settings/#configure-factory-preset-frequencies-for-abp-devices

#### What is missing? What do you want to see?
Add a note point in Adding devices documentation to configure `Factory Preset Frequencies` while creating an ABP device of non 64-channel band.

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?
No

benolayinka, 2021-08-05:
If @KrishnaIyer @adriansmares can provide more information about this, I can include it in https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/494

NicolasMrad, 2022-03-29:
is this still relevant?

KrishnaIyer, 2022-03-30:
I think it is. I'll pick this up this milestone.