TheThingsIndustries/lorawan-stack-docs #497: Document steps to claim a The Things Indoor Gateway via CLI

i3mSuperMan, 2021-08-03:
#### Summary
Currently, claiming of `The Things Indoor Gateway`(TTIG) via CLI is not documented.
#### Why do we need this?
Users can able to claim the TTIG via CLI.
#### What is already there? What do you see now?
The current documentation consists of claiming TTIG via the Console. Ref: https://www.thethingsindustries.com/docs/gateways/thethingsindoorgateway/
#### What is missing? What do you want to see?
Steps to claim a TTIG via CLI.
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No. @benolayinka

benolayinka, 2021-08-03:
@KrishnaIyer can you pick this up, or comment here with steps I can document?

KrishnaIyer, 2021-08-04:
I can take this issue.

closed 2021-08-09 by KrishnaIyer