TheThingsIndustries/lorawan-stack-docs #512: Document the entity already exists error in the getting started troubleshooting guide

i3mSuperMan, 2021-08-10:
#### Summary
The error `Entity already exists` was occurred due to the Email ID or User ID already registered in the tenant and seen when they were created and deleted by them already.

#### Why do we need this ?
To let users be aware of the User ID or Email ID already used.

#### What is already there? What do you see now?
Common issues and frequently asked questions by the users when getting started with The Things Stack. Ref: https://www.thethingsindustries.com/docs/getting-started/troubleshooting/

#### What is missing? What do you want to see?
Add `Entity already exists` error into the troubleshooting guide.

#### How do you propose to document this?
Insights:

- Let inform the users to check if the User ID or Email ID already exists in the tenant by login to the console from an Admin user account.
- Let note the Users that `IDs` are soft delete in the Identity Server. When a `User ID` or `Organization ID` is deleted, the `ID`  and associated `Email ID` are retained and cannot be reused within the tenant upon deletion, but it can be permanently deleted by the `Admin` user. Once, they are purged, they can be re-used to register a new entity.
- If a User ID or Email ID not found from the User Management section in the console, let the user to check and follow the below steps to re-register the user with the same `Email ID` and `User ID` after purging the deleted entries.
  - Login to the CLI with an account which has Admin privileges.
  - You can check the soft-deleted user IDs with the below CLI command and choose the User ID to purge. `ttn-lw-cli user list --deleted --all`
  - To purge use the below CLI command. `ttn-lw-cli user purge --user-id <User_ID>`
  - Re-register the user now.
  
  **Note:**
  - Please refer to [Installing the CLI](https://www.thethingsindustries.com/docs/getting-started/cli/installing-cli/) and [Login with the CLI](https://www.thethingsindustries.com/docs/getting-started/cli/login/) guides to get started with CLI.

#### Can you do this yourself and submit a Pull Request?
No. @benolayinka

htdvisser, 2021-08-10:
> When a User ID or Organization ID is deleted, the ID and associated Email ID are retained and cannot be reused within the tenant upon deletion, but it can be permanently deleted by the Admin user. Once, they are purged, they can be re-used to register a new entity.

The deleted user account can also be restored. I don't know if the Console supports this (@kschiffer / @bafonins ?) but the CLI definitely does.

kschiffer, 2021-08-10:
Restoring is not yet supported but will be via TheThingsNetwork/lorawan-stack#3585

closed 2021-08-19 by benolayinka