TheThingsIndustries/lorawan-stack-docs #515: Rename `AS_923_925_TTN_AU` to `AS_920_923_TTN_AU` in the frequency docs

i3mSuperMan, 2021-08-11:
#### Summary
Australia uses `AS_920_923` rather than `AS_923_925`. The documentation of the [frequency plans](https://www.thethingsindustries.com/docs/reference/frequency-plans/#as_923_925_ttn_au) needs to be updated (renaming from `AS_923_925_TTN_AU` to `AS_920_923_TTN_AU`) in line with the https://github.com/TheThingsNetwork/lorawan-frequency-plans/pull/27.
#### Why do we need this ?
Frequency plan naming should be in line with Australia's implementation.
#### What is already there? What do you see now?
[image]

#### What is missing? What do you want to see?
TTN Community Network frequency plan for Australia should be renamed to `Asia 920-923 MHz (used by TTN Australia)`
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
Yes.

i3mSuperMan, 2021-08-11:
It seems to be the frequency plans information in the documentation populated from the [lorawan-frequency-plans](https://github.com/TheThingsNetwork/lorawan-frequency-plans) repo. As we are not sure how to change it from the repo side,  we are assigning it to @benolayinka  here.

benolayinka, 2021-08-23:
Closed by https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/531

closed 2021-08-23 by benolayinka