TheThingsIndustries/lorawan-stack-docs #522: A section to show upcoming releases with tentative timelines to track `new features`.

ymgupta, 2021-08-16:
#### Summary
It would be great to make it clear to the TTS Users about the upcoming features and tentative ETA for the availability of them in V3.
#### Why do we need this ?

- Currently, we are suggesting to the users to track the features through the GitHub links. However, users are requesting the ETA of the new features.
- It would be easy for us to set their expectations on the timeline for releases, and it helps users find **the latest upcoming** in The Things Stack.

#### What is already there? What do you see now?

- We are tracking new features through individual GitHub issues.
- Also, there a `Release Notes` section for the current releases and past releases information. https://www.thethingsindustries.com/docs/whats-new/

#### What is missing? What do you want to see?
Consolidate upcoming features and their tentative ETA  in The Things Stack documentation.
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No. @benolayinka

benolayinka, 2021-08-30:
I don't think this is going to happen. @NicolasMrad you would be in charge of creating these timelines if so, what are your thoughts?

NicolasMrad, 2021-08-31:
Currently I don't think is a good idea (and it won't be that efficient). Users should just stick to checking our github for 3 main reasons:
1. When they check the issues on github, they can add their comment or add a new issue that they might think of that we haven't included it yet
2. Most of the times we push features to new milestones, or we modify them or just keep them in our current milestone just to keep an eye on them. Some critical features are decided before the release date by a week if they will implemented or not. So we can not commit to a 100% accurate ETA for a certain issue, there a lots of variables
3. Issues are in milestones, and milestones have a release date so users can know when a feature will be basically implemented from the release date.

closed 2021-08-31 by NicolasMrad