TheThingsIndustries/lorawan-stack-docs #545: Document the `user not found` error in the troubleshooting guide

ymgupta, 2021-08-28:
#### Summary
Add a troubleshooting section for User Management and document `user not found` error.

#### Why do we need this?
Adding common queries into the troubleshooting documentation will be useful for the Users.

#### What is already there? What do you see now?
No troubleshooting section for Users and Organizations. Ref:https://www.thethingsindustries.com/docs/getting-started/user-management/

#### What is missing? What do you want to see?
Add a troubleshooting section for User Management and address the user not found error.

#### How do you propose to document this?
**Insights:**
 > `user not found` error

The `user not found` error occurs when an unregistered user tries to log in to the TTS Console.
_Solution:_
- Cross-check the User ID of the TTS account.
- Contact Admin and check if the User ID is already listed in the deleted list. If yes, purge the User ID and register as a new user.

#### Can you do this yourself and submit a Pull Request?
No. @nejraselimovic

nejraselimovic, 2021-08-31:
@ymgupta may I ask how you got this error, i.e. can you tell me how to reproduce? I see you wrote that it occurs when an unregistered user tries to log in to the TTS Console, but:

- when I try to log into the Console using The Things ID together with an unregistered username, I get the `Invalid credentials` error
- when I try to log into the Console using credentials, i.e. an unregistered username, I get the `Incorrect password or user ID`. 

I used TTS Community Edition.

ymgupta, 2021-09-01:
@nejraselimovic,

> The `user not found` error occurs when an unregistered user tries to log in to the TTS Console.

Sorry, the above-reported cause is not valid. Please find the correct cause below.

> `user not found` error

The `user not found` error occurs when an unregistered user tries to `Reset the password` in the TTS Console.

closed 2021-09-07 by nejraselimovic