TheThingsIndustries/lorawan-stack-docs #579: Add a note that access keys in V2 cannot be used in V3

ymgupta, 2021-09-22:
#### Summary
In V2, we have Application access keys (only for ApplicationManager API) for Authentication and bearer tokens for authenticating to the gRPC and HTTP APIs. Whereas in The Things Stack, API keys are implemented for the simplest way of authorization. Hence, they can be used in all the HTTP & gRPC APIs. But, the access keys in V2 cannot be used in V3. So, it would be good to document this point in the [Major Changes In The Things Stack](https://www.thethingsindustries.com/docs/getting-started/migrating/major-changes/) docs.
#### Why do we need this?
To inform users that the V2 access keys cannot be used in V3.
#### What is already there? What do you see now?
The current documentation discusses major changes in The Things Stack (V3)
Ref: https://www.thethingsindustries.com/docs/getting-started/migrating/major-changes/
#### What is missing? What do you want to see?
Add a Note section [here](https://www.thethingsindustries.com/docs/getting-started/migrating/major-changes/) that the V2 access keys cannot be used in The Things Stack (V3).
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

nejraselimovic, 2021-09-29:
I'm not sure this is really relevant. Why would someone try using the V2 application access key to authorize something on V3? Is there a concrete case where users were confused by this?

We also have this in MQTT API details:

[image]

I think that from this it's evident that application access key and application API key aren't the same thing, and we don't mention the application access key anywhere in the documentation but here.

ymgupta, 2021-09-30:
> Is there a concrete case where users were confused by this?

@nejraselimovic, we have a query from one of the users, where they asked if V2 application access keys are still used in TTS V3, and if so, where to find them.

Consider closing this issue as we have come across this query from only one user till now. We will reopen this if it is still a necessity based on customer feedback.

nejraselimovic, 2021-09-30:
> > Is there a concrete case where users were confused by this?
> 
> @nejraselimovic, we have a query from one of the users, where they asked if V2 application access keys are still used in TTS V3, and if so, where to find them.
> 
> Consider closing this issue as we have come across this query from only one user till now. We will reopen this if it is still a necessity based on users feedback.

Yeah, let's close this for now and get back to it if we see the need.

closed 2021-09-30 by nejraselimovic